hrvach/deskhop #233: Failed to open device in Config Mode under Linux

dan-m8t, 2025-02-26:
Hi!

First let me thank you for this awesome project! I bought my deskhop from elecrow. I flashed the firmware as suggested and it worked out of the box. Then I wanted to configure monitor positions because output A is my Linux machine on the left, output B my windows work machine. Pressing left ctrl + right shift + c + o works fine under Linux, I can see the mounted drive with the config html file. I tried both Chromium and Chrome and get this error after pairing and clicking on "read":

```
Uncaught (in promise) NotAllowedError: Failed to open the device.
```
To configure it for my needs I just switched the outputs, so the USB drive gets mounted to my windows machine. This works like a charm, I was able to configure the deskhop as need but would like to configure it under my Linux machine.

Any hint?

OS: Arch
kernel: 6.13.4-zen1-1-zen

Google Chrome: Version 133.0.6943.141 (Official Build) (64-bit)

hrvach, 2025-02-27:
There is some guidance in the documentation to get the config to work:

A: You probably need to tweak /dev permissions or create a corresponding udev rules file and make sure your user is in the right group, like so:

/etc/udev/rules.d/99-deskhop.rules

KERNEL=="hidraw*", SUBSYSTEM=="hidraw", ATTRS{idVendor}=="1209", ATTRS{idProduct}=="c000", GROUP="plugdev", MODE="0660"

Let me know if it helps.

dan-m8t, 2025-02-27:
Thank you for the response. This didn't work for me sadly. Strange behaviour though: While in gaming mode and then hitting left ctrl + right shift + c + o for config mode the mouse just stopped working. Keyboard is fine, needed to exit gaming mode several times or replug usb  to get it to work again. I cannot see any message in dmesg or udevadm monitor that shows errors or something mh...besides that my Caps Lock LED isn't flashing nor working anymore while connected to DeskHop.

I use the 8bitdo Retro Keyboard with 2,4Ghz dongle attached to deskhop, probably this is the issue, but I don't have a usb c to a cable right now to test if this would solve this issue.

dan-m8t, 2025-03-01:
This fixed my issue! Thanks!

My mouse is still not working while in gaming mode and then entering config mode though, maybe this is a bug? Exiting gaming mode, then go back into config mode leaves the mouse working though. I will close this as fixed.

closed 2025-03-01 by dan-m8t