hrvach/deskhop #235: Add Support for More Monitors per Output, and Vertical Layouts

cdiddy1979, 2025-02-27:
I think the title says it all.  I like these configs, and it would be nice if deskhop supported it:

--------------------------------            --------------------------------
|    A    |    A    |    B    |    B    |              |    A    |    A    |    A    |    A    | 
--------------------------------             --------------------------------
|    A    |    A    |    B    |    B    |              |    B    |    B    |    B    |    B    | 
--------------------------------             --------------------------------

This next one sort of works today, but the cursor jumps screens unexpectedly sometimes.  This is not ergonomic, but its what fits on my desk right now:
----------------------------------------------------------------
|    A    |    A    |    A    |    A    |     A    |    B    |    B    |    B    | 
----------------------------------------------------------------

Faisalalali, 2025-03-13:
Damn, tell me about scalability...

hrvach, 2025-03-18:
Seems like adding the vertical switch axis is the most sought-after feature. I'll try working on this soon, if work permits.

Btw you're telling me you have 8 monitors/screens in front of you? Pics or it didn't happen! 😃  Are you working at air traffic control or I'm just getting really old and this is now what people use? 😄

cdiddy1979, 2025-03-19:
again, i want to stress that i know this is not the most ergonomic setup, i plan one day to remove the top/cabinet and get a monitor array-mount that stacks my monitors in a 4x2 arrangement... or just get a couple ultra-wides... or something.

[image]

alterwen, 2025-04-15:
I was going to ask for a "wrap-around" mode, in which any machine outer border switches to the other machine in the opposite border, kind of covering variety of situations