hrvach/deskhop #241: Doesn't work with dragging in blender.

xxredxpandaxx, 2025-03-11:
There is an issue when using this with the 3d software, blender. When you are dragging the mouse off the edge of the viewport in blender it is suppose to wrap around in that window. This obviously messes up when not in gaming mode. 

Would it be possible to have a preference setting to enable gaming mode on click? Or would this mess up other functionality somewhere?

[image]

Faisalalali, 2025-03-13:
I noticed a similar thing when using a streaming software like ParSec, it's like the mouse is moving to the bottom right corner when you drag something.

A quick workaround is by reverting the mouse back to the relative positioning by pressing `Left CTRL` + `Right Shift` + `G`

This will disable the smooth mouse transition between systems of course. So you could press `ctrl` + `caps lock` to switch between systems or just use this mode when working with your application.

I hope it is possible to fix this though.

xxredxpandaxx, 2025-03-13:
Ya I set up a macro button to swap between gaming mode and regular mode for this. The setting would still be super helpful since I use blender all day.

hrvach, 2025-03-18:
> Would it be possible to have a preference setting to enable gaming mode on click? 

Can you please elaborate a bit on this, I'm not 100% sure what you mean. A setting which would do relative mouse movements when dragging while holding the mouse button down? Or something like middle click to toggle relative mode? 

What is the desired behavior you are after? When the mouse goes off the edge, it should appear from the other side on the same screen as a wrap around thing? I'm grateful for the video, but please keep in mind I suck at Blender and don't even know if the video is showing the correct behavior or the wrong one :)

xxredxpandaxx, 2025-05-28:
This wouldn't work for this application. The cursor doesn't go all the way to the edge of the screen. It goes to the edge of the active window and then wraps around.