hrvach/deskhop #242: Double clicking in Mac OS

ronalith, 2025-03-13:
Hi! **Is there an option to set-up the double click speed in Deskhop settings?** My issue is that very often my clicks are registered as double-clicks instead of single ones in Mac OS. I've tried tweaking the mouse settings for faster double-click but it's useless. The only thing that really works is to use the mouse out of Deskhop... I use a Logitech MX Vertical. Here's some things I've tried:

- Switching USB cords
- Switching ports
- Wiping Deskhop config
- Wiping mouse settings in Mac OS
- Checking updates etc
- Changing double-click speed to the fastest in both Mouse and Accessibility settings
- Restarts of all kinds

Thanks!

ronalith, 2025-03-28:
Following up on this, I also tested two different mice to rule out a hardware issue. To reproduce: Have a bunch of folders in a list view in Finder. Start clicking at medium speed through different folders, not aiming to double click. The connection through deskhop will open a folder, eventually, while the same dongle connected straight to the mac will not.

ronalith, 2025-06-02:
Yeah, I've tried two different mice, different dongles, different cords, and lots of settings tweaked. I'm sort of learning to live with it at this point, and I guess there could be a chance that both of my mice aren't working like they did when purchased, but I also get no clicking issues using the same hub in two different Windows computers.

asif-rahman-sp, 2025-11-15:
also experiencing this and thought it was my G502 Lightspeed as they're known to have switches that go bad like that and double click, however I got a brand new one and lo and behold its still doing it, thats when I thought maybe its the deskhop and found this thread 😅 

can corroborate that i am hopping between two Mac OS's, mouse is a Logitech G502 Lightspeed as mentioned. I am using the deskhop to a usb hub on one side and a kvm on the other. Everything else works beautifully and I couldnt be happier, this double clicking is literally the only issue / inconvenience I am dealing with. 

I did want to add that there apparently is many threads out there with similar double click complains just specific to mac os, so I wonder if its not really desktop and just a macos thing? just to share in case it helps, i started using the `Linear` mouse app on macos and it has a debounce setting to restrict double clicks until a certain millsecond threshold between the clicks, hoping this workaround helps, I just started this today so ill try to report back later if anything..

hrvach, 2025-11-20:
I will try to get a Logitech mouse and recreate this, then dump the hid reports and see if it's doubled. Also implementing a debounce should not be hard, but I want to understand why it's happening first.

benkirton, 2025-12-17:
I'm using a Razer Cobra and I'm getting the double-click issue as well.

Using the event viewer in Karabiner Elements on macOS I don't see multiple click events on the clicks that trigger the double-click.

I set the speed to max, so manually triggering a double-click is much harder so I know I'm well below the speed to trigger it.

It appears to be tied to movement as a completely stationary series of clicks doesn't trigger it very often so I am assuming at this point when it does that I moved slightly without realising. Moving slightly as I click triggers it fairly reliably every 5-10 clicks.

I don't know where the logic is for the OS or if there is a different command something can send to trigger a dbl-c, but Karabiner shows the exact same thing each time:

```
  {
    "type": "down",
    "name": {"pointing_button":"button1"},
    "usagePage": "    9 (0x0009)",
    "usage": "    1 (0x0001)",
    "misc": ""
  },
  {
    "type": "up",
    "name": {"pointing_button":"button1"},
    "usagePage": "    9 (0x0009)",
    "usage": "    1 (0x0001)",
    "misc": ""
  },
```
 
Any pointers on where to look to see what is coming in to trigger a dbl-c on a single click?

Gwyki, 2026-02-03:
I am seeing similar issues, interestingly again with a Logitech G502 mouse and I unfortunately do not have another to test.

To be clear I am getting:
1. Spurious double clicks only within OSX (Tahoe 26.2). 
2. The double events happen reliably every 5-10 clicks or so.
3. This issue is **not** apparent in "game mode".
4. It happens regardless where the respective devices are connected suggesting not an issue with the UART routing.
5. Rebooting does not temporarily resolve it.

It seems very easy to see the issue when selecting different notes within the notes.app - this opens the note in a separate window which you almost surely do not want to do!

@hrvach do you have any suspicions, or somewhere to look first? Caveat: I don't have any experience with the Windows or OSX USB HID stack to understand why this is a particular issue on OSX. I also don't have a spare PICO to properly debug (is there an easy way to add a CDC to the composite device so I can at least pipe out mouse events?)

benkirton, 2026-02-04:
Back to add that I am now using a HyperX mouse and the issue is still present. I'm not sure if these mice have anything in common such as using the same specific something or other in the HID stack, or if it's all mice and the ones reported are just popular.

hrvach, 2026-02-04:
> I am seeing similar issues, interestingly again with a Logitech G502 mouse and I unfortunately do not have another to test.
>
> To be clear I am getting:
>
>     1. Spurious double clicks only within OSX [...]

There is a PR that I haven't merged yet because I'm not sure how it would affect other devices and OS (#317), but might help in this case. Want to try if the issue changes with that one? You can compile with the debug flag ON (check CMakeLists.txt near the top). That will give you a CDC device where you can dh_debug_printf() stuff.

kalarsen, 2026-06-27:
> > > Mind trying this attempt? (please note, OS _must_ be set to MACOS for the output the workaround is supposed to function on)
> > > [deskhop.uf2.zip](https://github.com/user-attachments/files/29399112/deskhop.uf2.zip)
> > 
Still not noticing much of a difference. Wish I had better news.