hrvach/deskhop #253: Multimedia Keys on Mac Keyboard A1243

macgaver, 2025-05-22:
New to this awesome software, everything work great but the multimedia keys on my mac keyboard are not working, they send the Function key equivalent. It's controlling a Mac and a Windows.

What can I do ? Thanks in advance

macgaver, 2025-05-30:
Is that keyboard working for someone ? Need to know if I am alone ;-)

hrvach, 2025-06-03:
Can you dump the HID descriptor for the keyboard so I can see how the media keys are defined and if there is something weird or special here? 

I unfortunately don't have such a keyboard so it's kinda hard to debug remotely regarding the key repetition issue.

macgaver, 2025-06-17:
Using winhiddump:

05AC:0200: Apple, Inc - Apple Keyboard
PATH:\\?\hid#vid_05ac&pid_0200&mi_01#8&21473a4&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
DESCRIPTOR:
  05  0C  09  01  A1  01  09  CD  15  00  25  01  35  00  45  01 
  65  00  55  00  75  01  95  01  81  06  09  B5  81  02  09  B6 
  81  02  09  B8  81  06  09  E2  81  06  09  EA  81  02  09  E9 
  81  02  81  03  C1  00  
  (54 bytes)
05AC:0200: Apple, Inc - Apple Keyboard
PATH:\\?\hid#vid_05ac&pid_0200&mi_00#8&2743f085&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
DESCRIPTOR:
  05  01  09  06  A1  01  05  07  19  E0  29  E7  15  00  25  01 
  35  00  45  01  65  00  55  00  75  01  95  08  81  02  95  30 
  81  03  05  FF  09  03  25  FF  45  00  75  08  95  01  81  02 
  05  08  19  01  29  05  25  01  45  01  75  01  95  05  91  02 
  95  03  91  03  C1  00  
  (70 bytes)