hrvach/deskhop #255: Issues with mouse and keyboard

B3voholic, 2025-06-01:
When plugging in directly to the computer the mouse and keyboard are recognized by their respective software.  When they are plugged into deskhop they aren't recognized at all by their softwares.   This makes it impossible to use the extra buttons on the mouse.  Would it be possible to somehow get deskhop to forward the information needed so the mouse buttons can be operational?

Gorillix, 2025-06-02:
@B3voholic,  I've had similar issues, and unfortunately, I don't believe Deskhop can passthrough the necessary information for specialized mouse/keyboard software to recognize devices directly.

For my Razer mouse, my workflow involves connecting it via Bluetooth directly to the computer to program the buttons using the Razer software. Once the macros/settings are saved to the mouse's onboard memory, I then switch it back to a wired/wireless connection via Deskhop. The programmed buttons typically remain operational this way.

You may need to adopt a similar workflow. Hopefully, your need for frequently changing macros isn't too high, as it can be a bit of an inconvenience.

hrvach, 2025-06-03:
I know this is probably not what you want to hear, but this behavior is by design and related to security. Any software that can connect to the kbd/mouse through the box has the potential to compromise that kbd/mouse and in turn do the same to the other computer, effectively "jumping the barrier".

 If you need to configure the device, hook it up directly while doing it. This most likely isn't needed very often, so it shouldn't be an issue. If it is, I think it's probably a niche use that isn't worth sacrificing security for. Apologies if you disagree!

Gorillix, 2025-08-05:
> I'm having the same problem with my Razer mouse.
> The thing is, I have a wireless mouse and it has a small RGB dock that I
> use to check the battery level of my mouse.
> So, with the box, not only does that not work, but also my automatic mode
> switch doesn't work (gaming mode with different polling and colors).
> I don't understand the security argument. The software would communicate
> directly with my mouse, and that's it.
>
> If it is by design, I assume there's a way to pass through the mouse and
> keyboard info to the computer (and vice versa).
> If that's the case, what about a configurable option that enables this?
> Whoever doesn't want to enable it, don't.

I would love a optional pass through mode if possible. Just not sure if the
solution would have the bandwidth for that or if that would break mouse
jiggle mode.