hrvach/deskhop #257: Allow RGB support

TheRedfoox, 2025-06-25:
Allow you to forward the RGB of the keyboard and mouse so that Synapse, iCUE and others can control the RGB

hrvach, 2025-08-17:
Apologies for the inconvenience. This will most likely not be fixed, as the security model of the device is strictly against any software on either PC connecting to the commonly-shared device (keyboard, mouse, etc).

TheRedfoox, 2025-08-25:
Simply to understand, is it a hardware constraint of the raspberry or is it a constraint given by the desired structure/direction of the Deskhop project?

As it is important for me unfortunately if it is the desired direction for the project I should start from scratch. But it will make one more open-source project on GitHub 😄