hrvach/deskhop #263: On one PC, mouse always jumps to top right corner after keyboard input

kevinlong206, 2025-07-16:
(accidentally closed this issue an could not re-open it, sorry for duplicate)

Hello,

I was wondering if this is a configuration issue or a bug. I am using two windows PCs with a single 4k monitor on each.

On just one of the PCs, whenever I type something, about 1 second later the mouse cursors jumps to the top right of the screen.

I have tried with the screensaver mode enabled and disabled.

What's strange about this is it's totally related to _keyboard_ input. If I wave the mouse around and pause, it doesn't jump. But if I type a single key, the mouse then jumps to top right of screen, every time, and only one one of my two PCs.

Interestingly, the issue seems to follow the USB port on my device (a "Desktop Jr" I purchased on Etsy). I initially though it must have to do with one of my PCs also having a touchscreen/stylus,  but I found that the problem follows the USB ports and *not* the PC. If I switch the ports, the problem happens on my other PC. Strange!

swaziloo, 2025-07-17:
This happens with my QMK keyboard (which has mouse enabled for a trackball) when switching with the "other" side, the receiving side jumps to the parking spot. That issue was first reported here: https://github.com/hrvach/deskhop/issues/151

Are you using a keyboard with mouse keys, or are you moving with the pointer connected to the keyboard side (not always the mouse side)?

swaziloo, 2025-07-17:
The exact behavior in my case.
Mouse attached to 'A' attached to Left Windows
Keyboard/trackball attached to 'B' attached to Right Windows

- Move cursor from Left to Right with mouse: mouse and trackball pointers are in sync (same position if I move either).
- Move cursor from Right to Left with mouse: mouse pointer tracks normally, trackball "homes" to the top-right corner.
- Move cursor from Right to Left with trackball: trackball pointer tracks normally, mouse "homes" to the top-right corner.
- Move cursor from Left to Right with the trackball: trackball and mouse pointers are in sync.

Let us know if you're seeing this same behavior.

kevinlong206, 2025-07-18:
So I opened this issue, but I'm no longer experiencing the problem.

I did change the "park position" to "previous", I also have "Force mouse boot mode", "Enable Acceleration" and "Enforce Ports" checked on. 

After changing some of these settings, I fully unplugged the my "Deskhop Jr" device for some time. When I came back and plugged it back in, it wasn't exhibiting the mouse jump issue anymore.

ak66666, 2025-07-18:
Thank you for sharing this solution.

For me it does not work, unfortunately yet not surprisingly, as my configuration/scenario is different.

hrvach, 2025-08-17:
To help me debug this, please be so kind to describe exactly what's connected to which side and dump all hid reports for all affected devices.

Most likely something else is also acting as a mouse, the box gets confused and parks the cursor. I don't know how to repro this, so you'll have to help with some diagnostic data if possible. Thank you!