hrvach/deskhop #274: Mouse speed on Linux with 2 screens

andrehil, 2025-08-14:
First of all, I wanted to say thanks for creating and sharing this project, it's amazing!

I'm having an issue on Linux (Kubuntu).
When I have 2 screens connected to this Linux machine, the horizontal speed is drastically higher than the vertical speed.
If I move the screens on top of each other, then the opposite happens, the vertical speed is much higher.

Also, I noticed one more thing.
This machine is configured to be on the left in Deskhop.
If I move my mouse to the left edge of the left screen, it goes back to the right screen.
That might be the culprit, but I'm not sure.

So, this is what happens when I move my mouse around

```
Linux machine -------------------------------------------------------     Mac machine --------------------
Left screen A <-> Right screen A <-> Left screen A <-> Right screen A <-> Left screen B <-> Right screen B
                                                       it starts here
```

If I disable one of the Linux screens, the mouse goes around the same screen if I move it to the left edge.

PS1: Deskhop is configured as Linux
PS2: This same machine is configured with dualboot, and this doesn't happen on Windows (though I need to configure Deskhop to Windows, otherwise it doesn't work properly)

j-f-d, 2025-12-02:
I don't think deskhop is designed to handle this use case. You could probably add code to deskhop, so that you could switch between two saved configs using a keystroke, depending on which monitor you have enabled, but there would be no way for deskhop to know that automatically.

andrehil, 2025-12-03:
Well, I just wish it worked on Linux the way it does on Mac and Windows.
For those, it doesn't matter if I have 1 or 2 screens; the mouse speed is always correct, no need to change the X or Y speed in the firmware config.