hrvach/deskhop #279: Is it possible to add a shared clipboard feature?It’s a really practical one

zhqsoft, 2025-08-30:
When I gave DeskHop to my colleagues to use, their first reaction was always "Can we get shared clipboard functionality?" It seems this is a really useful feature.
I noticed the chip's USB speed is only 1.5 Mbps, but that should be more than enough for sharing text via the clipboard.
For security reasons, maybe this feature could be made optional in the user settings, allowing users to decide whether to enable it themselves.
Thank you so much for your open-source project and all your hard work—hats off to you!

splinteredlight, 2025-09-02:
I would also like to see this!

Bunny-Head, 2025-09-09:
This is so funny: For weeks I have been thinking about this and I decided to open a request today just to find this 😄

hrvach, 2025-09-13:
It's not easy to implement without a dedicated helpers on both machines to communicate through the box. Having the ability to store and exfiltrate information from one computer to another would open a can of worms. It would be a convenient addition, I totally agree with that :)