hrvach/deskhop #282: Power (and possibly other system commands like sleep) only passing through on one side

cearka, 2025-09-02:
I have "Power" on my keyboard (my system is setup so hitting power doesn't shut down but just turns off the monitors).

The POWER definitely works on one machine (the one plugged into B), but on the other machine (the one on the A side).

I tried testing there button directly on both machines and it still works, some how its getting intercepted when it keyboard is in the deskhop.

closed 2025-09-07 by cearka

hrvach, 2025-09-13:
Try dumping what the power button actually does with usbhid-dump, I'll try to check what's blocking it.

cearka, 2025-09-14:
it ended up being an issues with they way the system control stuff was being passed.  had a buddy look at it and managed to fix it.