hrvach/deskhop #288: How to Enter Relative Mouse Mode Automatically via Code Instead of Manual Shortcut?

zoushiyin, 2025-09-20:
I need to press Ctrl+Shift+G manually to enter Relative Mouse Mode, but I'd like this to be automatically enabled at startup via code, without requiring manual input.

Is there a way to trigger Relative Mouse Mode programmatically on boot (e.g., by setting a flag or calling a function in main())?
Could you point me to the relevant part of the code where the Ctrl+Shift+G shortcut is handled? I'd like to simulate it during initialization if possible.
Additionally, I've noticed that when using a single board, the startup behavior is inconsistent:

About 50% of the time, the LED flashes three times and the device starts normally.
The other 50%, it fails to start properly (no LED flash or only one flash), requiring a reset.

zoushiyin, 2025-09-22:
@hrvach

hrvach, 2026-02-02:
> Looks like [#303](https://github.com/hrvach/deskhop/pull/303) adds an option to default to relative/game mode on startup.

yep, I need to test that one and try to perhaps add a relative mode switching. Was trying to investigate rp2350 usage possibility in the meantime, there seem to be some pio-usb issues I'm hitting with sdk 2.0...